arduino/lab-micropython-editor #18: Unable to find where to install packages

ProfessionalGooseChaser, 2022-11-10:
I'm using a a library not included in the packages that come with the IDE. I tried using 'pip install' in the terminal however it throws an error. There was nowhere else that I could find that would give me a usable terminal to install a package. 

I'm sure there's a certain way to format these Issues, but I'm newer to the industry and I wanted to ask for help when using your new product. Thank you so much!

(the package I was trying to instal was osascript)

ubidefeo, 2022-11-11:
@ProfessionalGooseChaser 
thank you for submitting this issue.

Micropython supports **root** (**/**) and **lib** folder to add modules.
They are findable by any script running based on their location on the board's file-system.

For now the only way is to copy your module to the local filesystem and from there copy the files manually

[image]

Hope that helps :)

ProfessionalGooseChaser, 2022-11-11:
Thank you so much for your help! I got the libraries I needed installed.

ubidefeo, 2022-11-14:
my pleasure 😃

closed 2022-11-14 by ubidefeo