arduino/lab-micropython-editor #19: Request: ARM build for Mac?

andypiper, 2022-11-11:
Recognising that this is early days, would it be possible to request that Mac builds target Apple Silicon as well as / ahead of Intel? thanks!

ubidefeo, 2022-11-14:
@andypiper 
Thank you for this, as an M1 user myself I want an `arm64` build but we're prioritising bugs and features.
Unfortunately GitHub does not support Apple Silicon runners, so it can't be automated but should be made ad-hoc on a private machine.
We plan on adding it to the build process for manually building from source 🙂

andypiper, 2022-11-14:
Understood! It was fairly straightforward to build locally. Thanks.

mikecee, 2024-03-12:
Apple Silicon runners [recently became available to open source projects](https://github.blog/changelog/2024-01-30-github-actions-introducing-the-new-m1-macos-runner-available-to-open-source/).  It seems pretty straight forward to _activate_ a Apple Silicon runner (single line change in `.github/workflows/build.yml` at `jobs > build > strategy > matrix > config`), but it looks like supporting both `x64` and `arm64` would need a tiny bit more thought. 

The simplest approach looks to be have two different runners (`macos-14` for Apple Silicon and `macos-13` for x64 are currently mutually exclusive w.r.t. Github runner availability), and remove the special-case forcing `electron-build` to only generate for x86. I've started a PR #95, looking for feedback from someone who understands the build process more than me.

ubidefeo, 2024-04-19:
Coming once PR #116 is merged to `development` and later `main`.
Expect 0.9.2 to have this in a few days, when we'll merge our latest fixes.

Thank you for the patience :)

Lapporatory992, 2024-05-18:
hi  is there a version for raspi 5? thanks

andypiper, 2024-05-20:
Looks like this issue is now resolved! Thanks.

closed 2024-05-20 by andypiper