arduino/lab-micropython-editor #22: Handle permission denied on serial port

logut, 2022-11-18:
A message should be shown if an user on Linux has no permissions on the serial port.

Currently the following error is thrown in dev console and nothing is happening in the UI.
`Uncaught (in promise) Error: Error: Permission denied, cannot open /dev/ttyUSB0    store.js:94`

ubidefeo, 2022-11-19:
Thank you for reporting this, @logut 
I'll make sure it gets handled 🙏🏼

kgbkgb, 2022-11-28:
Unable to connect on SUSE 42.1(x86_64). The USB works in uPyCRAFT, and is called /dev/ttyACM0 . Is the name of the USB port the problem?  Is it called /dev/ttyUSB0 as in comment above by "logut". The error in console is not about a port:
         gpu_memory_buffer_support_x11.cc(44)] dri3 extension not supported.

ubidefeo, 2024-04-19:
@logut @kgbkgb 

I'm doing some grooming on the issues, and since we already have 0.9.1 out I'd like to know if you still experience this.
The issue is pretty stale, I might close it next week if no feedback is given.
Thank you
u.

ubidefeo, 2024-04-26:
Closed after release 0.9.x

closed 2024-04-26 by ubidefeo