arduino/lab-micropython-editor #23: File type extension

Jeff-Squared, 2022-11-19:
At the moment it is possible to create a file without an extension, save it to disk then upload to the device where it becomes invisible to the user which takes up space in the flash. A new user might not realize this is happening until there is no room to upload a program, perhaps there is a way to force the entry of an extension before saving or uploading.

ubidefeo, 2022-11-19:
excellent idea, @Jeff-Squared 
maybe enforcing a regexp pattern might do the trick, but we need to handle pattern rejection and some sort of warning feedback on the field

Jeff-Squared, 2022-11-20:
I am using the app whenever I get the opportunity, I thought it might help a little if I let you know how it looked from a users point of view :o)

ubidefeo, 2024-04-19:
@Jeff-Squared 
it is now possible to create and list files without extension.

I'll close this issue
thanks
u.

closed 2024-04-19 by ubidefeo