arduino/lab-micropython-editor #24: Loading "boot.py"?

Kouzeru, 2022-11-20:
I've been using uPyCraft, and changes made mostly on boot.py
It's somehow not accessible in Lab MicroPython Editor

ubidefeo, 2022-11-21:
Are you using a recent version of MicroPython?
On which board?

ubidefeo, 2022-11-21:
@Kouzeru 
you may be running an old version of MicroPython.
I ignored the existence of uPyCraft and when I looked it up I saw it's over 5 years since the last commit.
Please make sure you're running a recent version of MicroPython.
We have tested with a wide variety of hardware and the issue you reported is not an issue.

You can provide further information about your bug, such as OS, versions, target hardware

Kouzeru, 2022-11-23:
I do have the latest MicroPython installed-

I tested the board furthermore, the files (including "boot.py") may occasionally shown on Lab Micropython Editor, and it disconnects when opening one of them. Sometimes all of files disappeared when reconnected while REPL is still working. Sometimes when trying to connect the board, the GUI is just hang on the port select menu. So I think it's mostly due to connectivity issues during filesystem access?

I'm using
Windows 10 21H1, 10.0.19043
Micropython 1.19.1 (2MB+ Flash)
Esp8266 (LOL1n Ver 0.1)

ubidefeo, 2022-12-14:
@Kouzeru 
just wanted to let you know that I'm keeping an eye on every issue and this one has not presented to me yet.
I'll keep my eyes open for now.
Maybe some other user experiences this and reports here

Auversss, 2023-01-13:
I have the same problem. I can open any file except boot.py.
Windows 11 22H2 22621.1105 
(sysname='esp32', nodename='esp32', release='1.19.1', version='v1.19.1-782-g699477d12 on 2022-12-20', machine='ESP32C3 module with ESP32C3')
lab-micropython-editor 0.6.0

ubidefeo, 2024-04-19:
I'd like to know if anyone still has this issue.

With it being stale, I might close it next week if no feedback is given.
Thank you
u.

ubidefeo, 2024-04-26:
Closed after release 0.9.x

closed 2024-04-26 by ubidefeo