arduino/lab-micropython-editor #25: Characters \r\n when loading a file are converted to line breaks in the file itself.

EugeneRymarev, 2022-11-23:
Problem 
[image]
Original
[image]
[File](https://github.com/robert-hh/FTP-Server-for-ESP8266-ESP32-and-PYBD/blob/master/uftpd.py)

ubidefeo, 2022-11-23:
@EugeneRymarev 
Please be more specific with describing your issue.

adding text and code blocks can better help us understand and compare.
Screenshots are not the right way to submit an issue

thank you

EugeneRymarev, 2022-11-23:
@ubidefeo 
[test.zip](https://github.com/arduino/lab-micropython-editor/files/10075812/test.zip)
Try to upload this file.
source
[image]

```
print('200 OK \r\n')

```

result after upload to device
[image]

```
print('200 
```

jcristine, 2022-11-26:
I am having the same issue.  The target file on the microcontroller, in my case a Pi Pico W, is truncated at the "\r\n" exactly as shown in the comment above this one.

EugeneRymarev, 2022-11-26:
@jcristine 
You can upload your files via ampy.

jcristine, 2022-11-27:
> "You can upload your files via ampy"

Yes - that is what I've been doing for years but was happy to see that the new Arduino IDE has it all in one UI (unless we use \r\n in code)

EugeneRymarev, 2022-11-28:
@jcristine, I meant that I would have to do this until this bug is fixed. By the way, the plugin for PyCharm has the [same bug](https://github.com/JetBrains/intellij-micropython/issues/212).

EugeneRymarev, 2022-11-29:
@ubidefeo, do you understand the problem?

murilopolese, 2022-12-01:
Thank you @EugeneRymarev for the report!
I think this is expected due to the way we are saving files on the board. 
The IDE (currently) send files to be board as text strings and I think this is the reason for the behaviour you reported.
I think we should look into this, @ubidefeo :+1:

murilopolese, 2022-12-05:
I did a quick investigation and I can reproduce the bug.
My new guess is that the method `read_until` from `micropython.js` may be related to the issue: 
https://github.com/arduino/lab-micropython-editor/blob/main/micropython.js#L72

chrisbuerginrogers, 2023-01-21:
I solved it by converting the desired string to binary and then writing it out as hex values:

gc.collect()
with open("Fred.py","wb") as bfile:
    f = bfile.write(bytes([0x73,0x22,0x10])).  #I do this in 256 byte chunks (so not these three random numbers)
    f = bfile.write(bytes([0x73,0x22,0x10]))
    f = bfile.write(bytes([0x73,0x22,0x10]))
    f = bfile.write(bytes([0x73,0x22,0x10]))

That seems to work fairly flawlessly

ubidefeo, 2024-04-19:
Fixed in 0.9.x

closed 2024-04-19 by ubidefeo