arduino/lab-micropython-editor #27: Apostrophes not buffered correctly in combination with special character in REPL

StevenCellist, 2022-12-14:
In most languages, characters like ```'``` and ```"``` are buffered until a second character has been typed.
The REPL however buffers these incorrectly: typing either of these two, followed by another special character, results in that special character rendered twice instead of rendering the apostrophe and special character.

Example: typing ```"]``` results in ```]]```, unless specifically entering ```␣``` after the apostrophe.

ubidefeo, 2023-01-03:
hey @StevenCellist 
I tried to figure out what you mean but to be honest haven't been able to understand

StevenCellist, 2023-01-03:
Hey @ubidefeo, thanks for following this up!
I hope I can be clear enough. 

An example: typing the following in the Terminal
```python
x["test"]
```
... results in ...
```python
x["test]]
```
This only happens on a keyboard-language which buffers the " for use in e.g. ä and ë.

An example is the Dutch (NL) keyboard.

Hope that helps!

ubidefeo, 2023-01-21:
hey @StevenCellist 

I cannot reproduce that one on Mac.
Even using combinations of ALT and '`' to add a modifier of following character it won't happen.
What OS are you on?

StevenCellist, 2023-01-24:
Windows; I have tried it on multiple machines and v0.6 has the same problem on my pc.

murilopolese, 2023-02-13:
I can't reproduce either. I'm on a linux machine with Swedish keyboard.

[image]

One thing to take in consideration is that the console/repl itself is not part of the editor. 
The source code and character parsing happens inside the board, by the micropython firmware.
All that the editor does is to pipe the keystrokes through serial exactly the way they are being captured by the javascript listeners.

I would recommend trying to type the same thing with another software such as `screen` or another serial monitor to see if the problem is with the javascript. If you are typing a combination of keys to input `"` the problem may be out of our reach :T

ubidefeo, 2023-06-16:
@StevenCellist 
we have completely reworked the escape/encode part of the save/upload workflows.
The bug will be fixed in upcoming `0.8.0`

ubidefeo, 2024-04-19:
Closed for non-reproducibility and staleness

closed 2024-04-19 by ubidefeo