arduino/lab-micropython-editor #31: Version 0.6.0 still reports as 0.5.0

dsssssssss9, 2023-01-05:
[image]

ubidefeo, 2023-01-08:
hi @dsssssssss9 
thank you so much for this

I made a rookie mistake there hardcoding the version.
Will change it to grab that off the package.json 🙏🏼 

Rest assured, though
 you are running 0.6.0 😬

ubidefeo, 2023-01-08:
@dsssssssss9 
I went to check the code-base and tested on my machine.
The correct version is reported 🤔 

Which OS are you running?

dsssssssss9, 2023-01-08:
Ubuntu 20.04 LTS

ubidefeo, 2023-01-08:
@murilopolese , can you check if the version is wrong on your Linux machine?

dsssssssss9, 2023-01-19:
any update on this please?

ubidefeo, 2023-02-06:
@dsssssssss9 
I have not managed to reproduce this issue on Mac or Linux.
I am using Ubuntu 22 on arm64

dsssssssss9, 2023-02-08:
closed as have completely wiped every trace of all versions & reinstalled from fresh copy of v6 from GitHub - now reports correct version

closed 2023-02-08 by dsssssssss9