arduino/lab-micropython-editor #33: If local folder is not chosen, locally saved files are nowhere to be found

ubidefeo, 2023-01-28:
Every time we launch the editor the Folder needs to be picked.
If this does not happen, choosing to create a file in the _Computer Storage_ creates the file in the editor, 
but when saving, although the operation is reported as successful, the file is not saved to any place the user can easily find.
I have run a search on my Mac and the file created and saved could not be found.

IMO there are two ways to approach this...

* Quick way
  * trigger to pick the Local Folder on save if it's not defined

* Best way
  * trigger to pick the Local Folder on save if it's not defined
  * save the chosen folder to persistent storage in a pref file

This has led to loss of files during my last teaching gig

murilopolese, 2023-02-06:
In my original idea, the save button should simply not be clickable if you haven't selected a folder.

There is also a feature idea for once the folder is picked to be remembered when you reopen the app. 
I think we should take this separately.

murilopolese, 2023-02-06:
https://github.com/arduino/lab-micropython-editor/pull/35

closed 2023-02-14 by murilopolese