arduino/lab-micropython-editor #34: When clicking "disconnect", the port remains open

ubidefeo, 2023-02-01:
Until the app is closed, the Serial port is busy.

To reproduce:
- connect to a board
- disconnect
- run `screen /dev/boardPortTTY 115200`

the message received will be 
`Cannot open line '/dev/boardPortTTY' for R/W: Resource busy`

Debugging the app did not bear any fruit.
The disconnection happens correctly when the **Connect** operation is triggered,
otherwise it would not connect to it, I guess.

murilopolese, 2023-02-06:
I could not reproduce on my Linux machine. What OS are you using?
Maybe it's not really disconnecting or there is a cache on OS level somehow.

ubidefeo, 2023-03-04:
Mac OS

Let's try and do a live session of debug soon.
I think the binding stays on, and on mac it just reconnects to the same resource

ubidefeo, 2023-06-16:
For anyone experiencing this, there's a fix and it will be part of 0.8.0

ubidefeo, 2024-04-19:
fixed

closed 2024-04-19 by ubidefeo