arduino/lab-micropython-editor #38: Upload/download buttons don't seem to work on mac-osx

icepick3000, 2023-03-10:
Somehow these buttons are always greyed out.. even when selecting a file on the left side. Getting a file to and from the board involved copy and pasting now. 

[image]

ubidefeo, 2023-04-16:
@icepick3000 
sorry for the late reply.
have you selected the local folder using the "Folder" button?
That path is not persistent, but I'm looking into saving a configuration file so that it remains and is retrieved at launch :)

jaenrig-ifx, 2023-04-21:
Hi there!

In my case, (using 0.7.0 windows) after properly selecting the folder, the files are uploaded but there are half-copied, corrupted, with missing parts...
Eventually, sometimes (don´t know the conditions for reproducing this issue) I also get the terminal output in the text editor?
[image]

I guess is a separate issue, but all starts when trying to copy something from the disk to the micropython device file system.

ubidefeo, 2023-06-14:
@jaenrig-ifx 
We've been exploring all these "missing blocks" issues when saving/uploading, but mostly on Mac.
We have some fixes that will be merged soon, but the Node SerialPort Mac binding seems to be problematic.
Next release will be more stable and include a series of blocking mechanisms that will prevent double sending of commands that can break the flow of data.
Bear with us, it'll get better 😃

jaenrig-ifx, 2023-06-15:
Thanks for the update @ubidefeo!

ubidefeo, 2024-04-19:
File corruption fixed as of 0.9 by moving to binary transfer of files.
Also files do not get immediately open at selection since the File Manager is in its own view

closed 2024-04-19 by ubidefeo