arduino/lab-micropython-editor #40: Version number not updated in about information banner

jaenrig-ifx, 2023-04-21:
After downloading the 0.7.0 release for windows, it seems the about version information is not properly updated:
[image]

per1234, 2023-04-21:
Thanks for your report @jaenrig-ifx! An equivalent problem was also reported for the previous release: https://github.com/arduino/lab-micropython-editor/issues/31
That issue was closed by the reporter, but the cause of the mismatched version data was never identified. So I'm not implying this issue is a duplicate; but only linking to relevant information.

per1234, 2023-04-21:
In this case it appears the problem is that the metadata was not bumped prior to the release:

https://github.com/arduino/lab-micropython-editor/blob/0.7.0/package.json#L4

https://github.com/arduino/lab-micropython-editor/blob/b15464b75990fa5e85e241eace00a1e4bc5e09cd/package.json#L4

It might be useful to document a release procedure. We found this very useful for the release managers in the Arduino IDE 2.x project:

https://github.com/arduino/arduino-ide/blob/main/docs/internal/release-procedure.md#release-procedure

ubidefeo, 2024-04-19:
We are paying more attention to the version number in the `package.json`,
but we'll definitely write a release procedure document.
This issue will be closed

closed 2024-04-19 by ubidefeo