arduino/lab-micropython-editor #46: Windows11 - Application disconnects port frequently

pbdaran, 2023-06-15:
OS: Windows 11
Application version: 0.6.0

When clicking on 'Run' icon, the port disconnects abruptly. The issue happens randomly and frequently.
After some counts of port disconnection, the application itself gets terminated.

ubidefeo, 2023-06-16:
hi @pbdaran 

could you provide the following?
* board model
* MicroPython version

The disconnection of a serial resource can be due to the board resetting itself for a number of reasons.
If you're working with an ESP* and there's a stack dump the board will reset and disconnect from the machine.
We have not experienced such issue so far, more info are welcome 🙏🏼

pbdaran, 2023-06-16:
I am using a **custom board** with ESP32-C3.
**Micropython**: [v1.20.0 (2023-04-26)](https://micropython.org/resources/firmware/esp32c3-20230426-v1.20.0.bin)

I monitored the reset signal of ESP32-C3 using oscilloscope. Didn't see any reset.
Also I didn't face the issue with Thonny IDE.

ubidefeo, 2023-06-22:
hi @pbdaran 

We have worked on a series of enhancements to make the application more sturdy.
I have had some issues with ESP32S2 which now are not there anymore in the main branch (we just merged today).

A new release is upcoming, just a little patience 🙏🏼

ubidefeo, 2023-08-17:
hi @pbdaran 

have you tried with our latest release?
I'm doing a bit of issues cleanup and if this is resolved I'd like to close it

thank you :)

pbdaran, 2023-08-18:
Hi @ubidefeo 
Yes, I am using the latest version for a while and I don't see the port disconnection issue.
Thank you 😊

closed 2024-04-19 by ubidefeo