arduino/lab-micropython-editor #47: Micropython for arduino nano

deljrf, 2023-06-19:
Hi, 
I'm unable to install micropython in my Arduino nano BLE board. 
Kindly help me in this regard

suggestions for any demo videos or tutorials will be helpful

ubidefeo, 2023-06-22:
hi @deljrf 

Installing MicroPython to the Nano 33 BLE is not super-simple.
Instructions can be found here
https://micropython.org/download/arduino_nano_33_ble_sense/

If you are not in a hurry I'd suggest you wait a little because we're coming out with something to help users in the process :)
🤫

deljrf, 2023-06-23:
Hi @ubidefeo 
I tried updating and uploading the boot and soft device through OpenMV IDE. After which I was able to code my Nano BLE sense in Arduino Lab.And the device is working fine with the micropython coding in Arduino Lab.I hope this process is okay for now

We would be very happy if there is any update for using the "Arduino lab" for "Arduino Nano 33 BLE sense or Arduino Nano 33 IoT".
Can't wait to try the new release. Please keep us in the loop for any updation on the same

Thank you so much for your response,

ubidefeo, 2023-06-23:
OpenMV is a way of installing the firmware, maybe just a bit odd to download an IDE just for that purpose but I'm happy you found it helpful :)

Keep an eye on this repo next week for an update
u.

deljrf, 2023-06-27:
Hi 
Are there any changes in Arduino lab? I'm not able to code my nano 33 ble sense controller in it. Unable to connect it through openmv IDE. I have uploaded the BLandSoftdevice in the Nano BLE, downloaded the .bin file for the controller. But not able to load the firmware in openmv IDE

Please help me resolve this

ubidefeo, 2023-08-17:
@deljrf 
sorry for the late reply, this being a "Lab" project we don't have a lot of time assigned to it.
Please download this tool to install MicroPython on your board, it should address all your needs :)

https://labs.arduino.cc/en/labs/micropython-installer

saurabh-2905, 2024-02-16:
> https://labs.arduino.cc/en/labs/micropython-installer

I tried using this to install Micropython on my Arduino nano 33 ble sense Rev2. It worked fine for the first time and I was able to run micropython scripts using Arduino-Lab IDE. However, the device got stuck while uploading a file, and then was not able to connect to the IDE. So I thought of reflashing the device. However, now it just fails to flash the firmware. In one of the attempts I got following error:

`Error running bossac: Command failed: "/Applications/MicroPython Installer.app/Contents/Resources/app/node_modules/firmware-flash/bin/darwin/bossac" -d --port=/dev/tty.usbmodem1 -U -i -e -w "/Applications/MicroPython Installer.app/Contents/Resources/app/node_modules/firmware-flash/bin/firmware/SoftDeviceUpdater.bin" -R`

Later I also tried running the Arduino scripts, which work just fine. But I am not able to flash micropython since then. I also tried using BOSSAC in terminal. It says the firmware is installed but later the device is not seen on the serial port. Same happens with OpenMV installation. It is able to install micropython successfully, but then device is not found. 

Can you please help me with this? Thanks.

ubidefeo, 2024-04-19:
@saurabh-2905 @deljrf 
This should be reported to the Arduino MicroPython Installer's repo
https://github.com/arduino/lab-micropython-installer

This issue will be closed