arduino/lab-micropython-editor #60: Can't add new code into file after moving or loading it

FOXimir, 2023-08-24:
Hello all,

I am not able to insert new code into my file after loading it into Arduino Lab editor or copy file between Arduino and my computer through Arduino Lab.
Cursor in editor is blinking but can't insert new character. I only can delete characters by using Backspace and Delete keys.
It is not working for both my locally saved files as well as files saved in Arduino memory.

### Example:
Loading main.py from Arduino memory into editor:
[image]
After loading I can move cursor and delete characters but can't insert new one.
[image]

### Workarounds:
1. Close Arduino Lab app and open it again, first time loading seems to be working fine.
or
2. Mouse click outside of Arduino Lab app and then click back to the editor.

### My configuration:
Windows 11 Pro 22H2
Arduino Lab for MicroPython 0.8.0
Arduino Nano ESP32 
MicroPython v1.20.0-384-g2919a9fbf

AlexinHolland, 2023-11-06:
I am experiencing exactly the same symptoms as mentioned above. Moreover, I notice that a script such as boot.py stops when I want to be able to view the files on the Nano ESP32 board and in my PC directory via the “Files” button. I then have to restart the script 'boot.py' with the 'Run' button. When I then “dis-Connect” the board, the name changes from “boot.py” to “undefined” and I can only make changes again by doing an “out-off window-click” and then choosing for the “ boot.py” script to edit it. These actions make it very difficult for me as a beginner of “Arduino Lab for MicroPython” and give me an insecure feeling of a reliable programming tool.

Using windows 10 
Arduino lab 0.8.0
Micropython V1.20

ubidefeo, 2024-04-19:
Please test version 0.9.1 and report if this still happens.
I had it happen to a couple students of mine during class, also with preview versions of 0.9, 
but we don't have a sure path to reproducing the bug.
If anyone can help us track it down with some reproducible steps we'll be able to look into it.

Thank you

ubidefeo, 2024-12-17:
closed for lack of feedback

closed 2024-12-17 by ubidefeo