arduino/lab-micropython-editor #61: 'Disconnected' banner

AppsByDavideV, 2023-09-05:
Hello,
it would be nice to remove the  yellow 'Disconnected' banner that hides files, so it will be possible to edit them on local folder  without a connected device.

Thanks

ubidefeo, 2023-09-06:
hi @AppsByDavideV 

Will definitely put this one in the backlog because it can be pretty annoying 😄 

Thank you for the suggestion
u.

murilopolese, 2023-09-06:
Thank you for your suggestion @AppsByDavideV ! We'll have a look into it. 
As a work around, as soon as you save a local file the banner will disappear.

IvanR3D, 2024-03-24:
> Hello, it would be nice to remove the yellow 'Disconnected' banner that hides files, so it will be possible to edit them on local folder without a connected device.
> 
> Thanks

It is easy to solve this by increasing the z-index of #panel.open (in the theme.css file) to 10000 (ten thousand). Then the file panel will get over the disconnected message that currently have a z-index of 9999.

ubidefeo, 2024-04-19:
@AppsByDavideV @IvanR3D 
version 0.9 has a rewritten UI which does not exhibit these annoyances.

Please try it and let us know

closed 2024-04-19 by ubidefeo

AppsByDavideV, 2024-04-26:
@ubidefeo : I see, nice UI, great!