arduino/lab-micropython-editor #63: Connected device disconnect frequently while uploading files

AppsByDavideV, 2023-09-09:
Hello,
as per title, micropython device suddenly disconnect while uploading files at whatsoever  upload percenage .
No matter the file size, it can be few bytes to kbyte,
The worst thing is that, even if the banner says 'uploaded', the file in most cases is corrupted.

Device Esp32 Wrover with SPI ram, mpy 1.19 and 1.20 . Onboard serial chip is CH340.
PC win10 and win 11, no matter.
Arduino lab 0.8.0

I must add that with some other IDE (i.e. ESP32 MPY jama) it happens rarely and upload of files is way faster (almost 10x).
But i prefer this one and fixing the problem would be great.

Thanks.
\Davide

ubidefeo, 2024-04-19:
@AppsByDavideV 
version 0.9 has much faster upload speeds and this kind of disconnections should not be happening.
Please also make sure you have more recent versions of ESP32 MicroPython because around version 1.20 there was a bug on the S3 and C3 which caused some file-system issues when saving data

AppsByDavideV, 2024-04-26:
Hi @ubidefeo : I tried it, but uploading of files still hangs during upload. Processor ESP32 WROOM (no S3 or C3). Anyway this is a nice update! Keep up the good work!

ubidefeo, 2024-04-29:
@AppsByDavideV 
thank you :)

have you tried with other devices?
the disconnection might even be due to the board browning out under load, might be an electrical issue.
if you have another board exactly like that and give it a test, it might confirm or deny the issue and we'd investigate more deeply.
could you, by any chance, make a zip of the files you're trying to copy and provide it to  me?

AppsByDavideV, 2024-05-01:
Hi @ubidefeo , my application requires a lot of ram, that's why I'm using ESP32 with SPI ram. 
Anyway, I gave it a try with different files and different esp32 module + usb cable (maybe):
- PC OS Win11
- Lab-micropython-editor 0.10.0 
- Processor ESP32WROVER 4M flash + 4M SPI ram from TTGO - USB type C connection
- Mpy 1.22 clean install
- Files: [Drive, password protected](https://drive.google.com/drive/folders/1a0hfLhqEx8PRfePNVUkxToiMa7xOXthk?usp=sharing) , please send an email to support <@>appsbydavidev.it  for password.
- 1st test using this IDE: 
**All files selected for upload in one pass**. Slow upload , blocked at 59%, got disconnection, see left top icon in screenshot- disconnected and reconnected the chip, got the file list on it,  tried to open an uploaded folder: hang. Must close the IDE.
- 2nd test with Thonny on same, cleaned chip: succesfully uploaded all files.
- I must add that uploading with Thonny is still a lot faster .

Attached some screenshots:
1: files list on both sides
2: upload blocked at 59%
3: can't access folders on chip

Hope it helps !
Bye
[image]
[image]
[image]

ubidefeo, 2024-12-17:
closed for further feedback

closed 2024-12-17 by ubidefeo