arduino/lab-micropython-editor #64: Loading forever when trying to connect esp32-c3

ningelsohn, 2023-09-17:
Using macOS Ventura 13.1 and Arduino Lab 0.8.0 I get an infinite loading wheel when trying to connect a XIAO ESP32-C3.
I can't find a list of supported devices but in any case the action should either lead to a successful connection or fail (right?).

ubidefeo, 2023-11-23:
@ningelsohn 
some custom builds of MicroPython have a different REPL response and this editor is not able to parse it.
can you try flashing another ESP32-c3 build from here and see if you can connect?
https://micropython.org/download/?mcu=esp32c3

ubidefeo, 2023-12-08:
@benj5378 
do you mean that now you can connect using the Lab for MicroPython Editor?
which board and which build of MicroPython?

I'd love to setup a test and see if I can spot where it breaks

Thank you :)

benj5378, 2023-12-09:
Loading forever happens as well when trying to save file to the Arduino Nano ESP32 - not sure if this is due to permission or just because it got disconnected. Other than throwing a permissions error, the communication line should also have a general timeout - this would stop infinite loads whatever the reason might be.

@ubidefeo Sorry to disappoint, but there seems to be quite some work needed on this application for even doing simple programs. Some very uncool UI stuff:
- Until the script is given another name than `undefined`, it's not possible to save. In other words, you could also say that the save button is very unintuitive, what is needed for enabling it.
- File manager, that can't create folders? This is an important feature that should be very easy to make, currently needing to do `os.mkdir(path)` manually.
- Files can't be saved onto the computer? What? Or is it the save procedure that is very unintuitive
- It's not possible to create a new file? Unless first creating one in the file manager? Should be possible to create new unsaved files.
- It's not possible to move folders using the file manager?
- It's not possible to create new files, move files or delete files as long as the default opened `undefined` file isn't saved?
- Sometimes when pressing the run button, some bug happens, where nothing happens. (No `OK` in the terminal)
- I don't know, if this is ESP32 or IDE side thing, but perhaps there should be an automatic newline after the `OK` often given, instead of the first `print(msg)` being on the same line as the `OK`
- Tabs for multiple files opened at once is needed - but I guess you're aware of that already
- It should be possible chagning the editor font size without chaing the size of the whole gui (meaning buttons too).
- The controller name `Arduino Nano ESP32` could preferably be shown instead of just `/dev/ttyACM0` in my case

ubidefeo, 2024-04-19:
@benj5378 
please download version 0.9.x and most of your disappointments will be made up for :) 

closing the issue since it is no longer relevant but attributed to 0.8 problems and shortcomings

closed 2024-04-19 by ubidefeo

0xifis, 2024-08-05:
> @benj5378 do you mean that now you can connect using the Lab for MicroPython Editor? which board and which build of MicroPython?
> 
> I'd love to setup a test and see if I can spot where it breaks
> 
> Thank you :)

I encountered a similar issue with connection.

OS: MacOS 14.4.1
Lab version: 0.10
Board: ESP-32 DEVKITV1 (WROOM)
Micropython: [v1.23.0 (2024-06-02)](https://micropython.org/resources/firmware/ESP32_GENERIC-20240602-v1.23.0.bin) **Did not work**
[v1.22.2 (2024-02-22) .bin](https://micropython.org/resources/firmware/ESP32_GENERIC-20240222-v1.22.2.bin) **Worked**

@ubidefeo Great work pal, appreciate putting in the time to develop this side project, although i feel that it is already in a good enough state for many to have their first micropython experience with