arduino/lab-micropython-editor #68: Dark Mode

mattveraldi, 2023-12-31:
Hi,

I like this project! I was wondering if there are any plans to add a dark mode switch — my eyes are sensitive to bright colors.

mattveraldi, 2023-12-31:
I am implementing it myself, if i will manage to do it I will be happy to share it :)

ubidefeo, 2024-01-06:
hi @mattveraldi 
Thank you for the interest :) 
This being a Lab project it's still getting the time it finds, but we'll be happy to get a PR and merge it if it brings advantages.
Unfortunately I don't see this getting a theme management system, but I'm sure a user could get hacky with themes since it's all in the style files.

I'm gonna keep an eye on this, please feel free to share your changes and I'll look into them.
Also keep in mind that we have another version in the works, which we're not building because it's not even ready for alpha, but it changes the design quite a lot :)

ubidefeo, 2024-04-19:
@mattveraldi 
we have a new version, 0.9, which changes the UI entirely.

Since theming is not on our list of enhancements, but we'll eventually look into it for accessibility reasons, you can play with colours on your own installation, but we'll not be able to accept a PR on something not yet structured to accept these options.
Thank you for being propositive, though, we really appreciate it :) 

I will be closing the issue as part of our internal grooming

closed 2024-04-19 by ubidefeo