arduino/lab-micropython-editor #101: Rest button in IDE

stevejoung, 2024-04-06:
The reset button in the IDE doesn't reset the board like removing the USB and replugging it does.

ubidefeo, 2024-04-09:
hi @stevejoung 

The reset in the interface is a MicroPython `machine.soft_reset()`.
Unplugging the board would result in a disconnection from the editor and the REPL

It is intended to be so :)

stevejoung, 2024-04-10:
My problem is that any LEDs lit by a previous problem are not extingushed on reset, making futher use of the LEDs for status indication impossible.

ubidefeo, 2024-04-19:
@stevejoung 

from the [MicroPython documentation](https://docs.micropython.org/en/latest/wipy/tutorial/reset.html)

> A soft reset simply clears the state of the MicroPython virtual machine, but leaves hardware peripherals unaffected. To do a soft reset, simply press Ctrl+D on the REPL, or within a script do:
> ```
> import sys
> sys.exit()
> ```

To obtain the results you expect, you should issue a hard reset.

> A hard reset is the same as performing a power cycle to the board. In order to hard reset the WiPy, press the switch on the board or:
> ```
> import machine
> machine.reset()
> ```

A hard reset would disconnect the board from the editor, which is not what we want/need.
If a disconnection is acceptable for you, you should hard reset the board using its onboard reset button

closed 2024-12-17 by ubidefeo