arduino/lab-micropython-editor #126: UI frozen with spinning wheel saying 'loading files'

AppsByDavideV, 2024-04-30:
Hello devs,
as per title, interface frozen, lab-micropython-editor version 0.10.0.
How: after the first start where I chose the working files path, I changed the name of the working folders on the PC. Of course the app cannot find the new files path. 
The expected behaviour is to ask for a new path if the previous is invalid, but the software is blocked instead.
[image]

Cannot find where that information is stored.

Bye

ubidefeo, 2024-05-01:
thank you for reporting this, I think it escaped our attention :D 

you can fix it by deleting the settings folder in 
`C:\Users\{USER_NAME}\AppData\Roaming`

secretsather, 2024-07-14:
Same isue here. I understand the fix (for now) but hope someone is working towards an automatic solution.

ubidefeo, 2024-12-17:
Fixed in 0.11.0
https://github.com/arduino/lab-micropython-editor/releases/tag/0.11.0

closed 2024-12-17 by ubidefeo