arduino/lab-micropython-editor #127: stuck on connecting

mindoindo, 2024-06-14:
[image]
I connected my nano esp32 and clicked on connect then com5 and its been stuck on connecting ever since

pmossel, 2024-06-20:
Same issue, connecting my new alvik, the IDE is stuck in "Loading files."

ubidefeo, 2024-07-10:
We have found a few users who have corrupted file systems after updating MicroPython.
Have you performed such operations before this issue appeared?
Which board and version of MicroPython are you using?

akanksha444, 2024-07-11:
I am facing the same issue with Nano ESP32

mrdowling, 2024-07-16:
Arduino Nano ESP32

ubidefeo, 2024-12-17:
Fixed in 0.11.0
https://github.com/arduino/lab-micropython-editor/releases/tag/0.11.0