arduino/lab-micropython-editor #128: MAC OS: UI frozen with spinning wheel saying 'loading files'

mrdowling, 2024-07-15:
I have been using Arduino Lab for MicroPython for weeks without issue. I was adding some new program files to my file location. Today the it is stuck in the loading.
[image]

ubidefeo, 2024-07-16:
please refer to the answer in #127 
we found out that some users end up corrupting their file-system and the editor cannot successfully test for the `lib` folder.
we need a bit of background work to address timeouts on operations

mrdowling, 2024-07-16:
Arduino Nano ESP32

mrdowling, 2024-07-16:
@ubidefeo Thanks for your help. What background information do you need?

I am running a laptop with MAC OS for the Alvik robot with an Arduino Nano ESP32.

mrdowling, 2024-07-17:
hanks for your help. What background information do you need?

I am running a laptop with MAC OS for the Alvik robot with an Arduino Nano
ESP32.

gbr1, 2024-07-19:
#130 

spammed also on Arduino_Alvik library 2 times

gbr1, 2024-08-06:
@mrdowling  I had a similar issue because I deleted a folder that was opened by lab for micropython.
I solved by deleting the `~/Library/Application Support/Arduino Lab For Micropython` folder

anomaleo, 2024-08-08:
@gbr deleting the ~/Library/Application Support/Arduino Lab For MicroPython folder does not resolve the issue for us with MacOS Sonoma 14.6.1 and MacOs Sequoia Apple Silicon M1, M2, M3 arm64 builds Version 0.10.0.

Cloned & Built from source for Version 0.9.1 for arm64 - resolves the issue.
Is there a patch for 0.10.0 in the works ?

ubidefeo, 2025-04-12:
No feedback from OP, but issue is considered resolved

closed 2025-04-12 by ubidefeo