arduino/lab-micropython-editor #130: Arduino Lab for MicroPython stuck in loading MAC OS

mrdowling, 2024-07-17:
I have been using Arduino Lab for MicroPython for weeks without issue. I was adding some new program files to my file location. Now the app is stuck in the loading upon startup. I'm using an Alvik robot with an Arduino Nano ESP32. My laptop uses MAC OS.
[image]

ubidefeo, 2024-12-17:
Should be fixed in 0.11.0
https://github.com/arduino/lab-micropython-editor/releases/tag/0.11.0

Real-John-Cheung, 2025-02-13:
I am still encountering this issue, in both the Mac app (v 0.11.1 on macOS 15.3, with a M4 Max chip) and Chrome (same machine Chrome version 133.0.6943.99 (arm64)). 

I am using a third-party board with Micropython firmware, not sure if that's why it doesn't work. The serial does connect as when I use the reset button on the board I can see the serial information updating behind the frozen UI, which is quite funny. 

If I use the file upload tab in the Mac app it connects with no issue, but if I try to connect to the board on the code tab the UI just freezes, which is confusing.

[image]

ubidefeo, 2025-04-12:
hi @Real-John-Cheung 

sorry for the late reply, we're going through a Lab sprint so I only saw your reply now.
Your original issue was related to the old version of the editor trying to create a `lib` folder in root, which is not possible because `/` is not writable, while `/flash` is.
Can you tell me which board you are using?

I'm working on a fix to allow auto-recognition of the board's navigation root, which in your case should be `/flash`, so I guess there's an attempt at reading/writing the root and that makes it hang, but it should not behave like that.
One of the boards I test with does have `/flash` since it's in the Renesas port.

The more info I get from you the better, and thank you for already being detailed :)

sebromero, 2025-05-06:
Closing this due to inactivity. Should be resolved.

closed 2025-05-06 by sebromero