arduino/lab-micropython-editor #135: Stuck in loading files

genx0521, 2024-09-15:
[image]

Hello, 

I have been stuck loading files while opening the Arduino lab for micropython. 

It was working fine and could work with it without a problem. I'm not sure what happened between it working and not working. I have deleted the folder and unzipped a fresh version downloaded and still the same issue. I have tried previous versions as well and still have the same issue. 

Any insight would be great.
;-)

genx0521, 2024-09-15:
Solved the problem - I opened the application as admin and it worked. Even though I am already the admin of my machine.

devcylenz, 2024-10-29:
Had the same problem, the issue is that the tool saves the folder for local files, and if that folder is moved, renamed or deleted, the program is stuck on startup.

devcylenz, 2024-11-01:
The workaround for now is (at least for me it was) to open the devtools inside the arduino lab and check the console. There you can find a warning that some folder can't be found. Create that folder (or folders) so it exists again. Restart the app and you should be good to go.

genx0521, 2024-11-01:
Another solution that worked for me was;  go into %appdata% and find the arduino micropython folder and delete it and restart the application.

ubidefeo, 2024-11-03:
hey folks

@genx0521 's is for now the best solution
>Another solution that worked for me was; go into %appdata% and find the arduino micropython folder and delete it and restart the application.

on Windows the app stores some extra data which it expects at start, but we need to look at why it does so.
Thank you for your patience, we'll look into it

gkneen, 2024-11-05:
To fix this on Linux Mint 22
1. Open the Home folder, right click and choose "show hidden files"
2.  Open the .config folder
3.  Right click on the folder "Arduino Lab for Micropython" and choose "rename"
4. I renamed mine to "Arduino Lab for Micropython.old"
5. Restart the app. It should now ask for  location to save micropython files
6. Delete the .old folder if everything works

ubidefeo, 2024-11-06:
thanks everyone

a fix has been merged, we'll make a new release in a few days 🙏🏼

ubidefeo, 2024-12-17:
Fixed in 0.11.0
https://github.com/arduino/lab-micropython-editor/releases/tag/0.11.0

closed 2024-12-17 by ubidefeo