arduino/lab-micropython-editor #141: Insert cursor dissappers in editor

breivog, 2024-11-17:
Normally when using the editor , you can move you cursor over a word, then click, and get an insert cursor to appear at that spot. Then you can enter new text, copy, delete, etc.    However, if you now go to the files manager and delete a file (the trash can),  the insert cursor now no longer appears when you click over a word.  If you restart the editor, then it works again.

ubidefeo, 2024-11-17:
thank you for reporting this, @breivog 

getting the steps to reproduce can be helpful to find a solution.
I think I have seen this happen on Windows without being able to replicate it.

Which OS are you running?

ubidefeo, 2024-12-04:
@breivog 

we have fixed this bug and it will be resolved in version 0.11.0, coming out by next week

thank you for your patience

ubidefeo, 2024-12-17:
Fixed in 0.11.0
https://github.com/arduino/lab-micropython-editor/releases/tag/0.11.0

closed 2024-12-17 by ubidefeo