arduino/lab-micropython-editor #177: Arduino lab doen,t find alvik files

johanverheyden, 2025-03-08:
Alvik is connected and is found by p.ex Alvik-update : this works fine.
However Arduino Lab doesn't show the . py files on the Alvik, although Thonny does !
Arduino Lab only shows the .py files on my Mac and neither can I transfer these files from the Mac to the Alvik, Thonny does.
I tried the web version of Arduino Lab and ... there my Alvik files are shown.
What could be the problem?

ubidefeo, 2025-03-09:
hi @johanverheyden 

on which OS are you?
have you made significant changes to your Alvik's Nano board's file system?
at the moment Lab for MicroPython has trouble parsing file names with non-standard characters such as accents or extended characters.

Could that be your issue?
Maybe a file has some "broken" name?

We know this is a bug we need to address

johanverheyden, 2025-03-09:
> ubidefeo

Good morning, thank you for your answer.

I am on Mac Apple M3 pro OS Sequoia 15.3.1.
I am currently following the lessons on the Arduino.cc <http://arduino.cc/> , and just started the “ moving” chapter.
I named my files as was indicated in the book ( see the screen view )

Everything went perfect and then, suddenly, there were no more files shown on the “ left side” of the screen.

I searched for solutions, and also grounded the B1 trying rebooting Arduino Nano Esp32 manually, but nothing succeeded, but this was, like I said, after the incident occurred.

I hope that you can help me.

Have a nice day !

Johan

johanverheyden, 2025-03-09:
Now with the screen view.
￼

ubidefeo, 2025-03-11:
hey @johanverheyden 

did you attach screenshots?
I think you replied via email and the attachments got lost

closed 2025-03-12 by johanverheyden