arduino/lab-micropython-editor #178: [Bug Report] Chinese text displays garbled characters in editor interface

sunny-zls, 2025-03-09:
Description:
Current version exhibits encoding issues when displaying Chinese characters (and potentially other non-ASCII characters) in the editor interface. This affects both code comments and UI elements.

Suggested Solution:
1. Implement proper UTF-8 encoding support throughout the application
2. Add language selection option in settings
3. Provide encoding format configuration for file operations
4. Create localization framework for future multilingual expansion

ubidefeo, 2025-03-09:
thank you for your issue, @sunny-zls 

we know there's an issue with characters encoding, I think this was due to MicroPython itself not supporting extended characters in the file system.

some of our students had this issue too when naming files with accents.
definitely gonna investigate this

about your suggesitons, I'm fairly sure point 4 won't be addressed anytime soon.
This is still a project we work on as a Lab activity within Arduino.

There's an upcoming version with some enhancements but wether or not a fix for this issue makes it through is still up in the air.
Definitely marking it as bug, so thank you for pushing :)

ubidefeo, 2025-08-21:
Addressed in #193 
Fix available in next release

closed 2025-09-02 by ubidefeo