arduino/lab-micropython-editor #179: [Feature Request] Integrate file browser and editor in single-view interface

sunny-zls, 2025-03-09:
Description:
The current separated interface for file browsing and code editing requires excessive context switching when working with multiple files. This impacts workflow efficiency.

Proposed Implementation:
1. Create a split-view layout with collapsible file browser panel
2. Maintain persistent editor workspace while browsing device files
3. Implement tabbed interface for multiple file editing
4. Add keyboard shortcuts for rapid file navigation

ubidefeo, 2025-03-09:
hey @sunny-zls 

thank you for this report and proposed solutions.

>Proposed Implementation:
> 
> * Create a split-view layout with collapsible file browser panel
Initially this was the layout, but it proved to be too crammed to make it really usable.

> Maintain persistent editor workspace while browsing device files
We also explored a sidebar design but for now the 2 context makes it clearer and prevents some UI issues we bumped into
Is it what you're referring to?

> Implement tabbed interface for multiple file editing
We support multiple file editing with tabs

> Add keyboard shortcuts for rapid file navigation
This is coming in the next release, with shortcuts for quickly switching between the views

ubidefeo, 2025-04-12:
No feedback from OP, but issue is considered resolved/obsolete

closed 2025-04-12 by ubidefeo