arduino/lab-micropython-editor #181: Files wont sjow in "Files"

tdh79, 2025-03-23:
No files are shown, have had some problems transferring files but now it wont show anything on the left side.
Works fine otherwise

ubidefeo, 2025-04-11:
you may have a broken file name.
I have seen this happen with accents or special characters in file names.
It ends up breaking the returned list of files and things start acting funny.

it's a bug we're aware of, and the way to solve it would be for you to use REPL to list files and delete or rename the offending ones

```python
import os
os.listdir()
os.remove('file_with_bad_name.py')
```

if the bad name is a folder make sure it's empty before you can delete it

ubidefeo, 2025-05-14:
Closed as stale and unanswered

closed 2025-05-14 by ubidefeo