arduino/lab-micropython-editor #182: Please add Dark Mode

jim201414, 2025-03-26:
Please add **dark mode** because the light mode **burns my eyes**.

-----------------------------------
**THE SETTINGS POV:**

> LIGHT_MODE: **_ENABLED FOREVER_**
> ����_MODE:  **_DISABLED_**

ubidefeo, 2025-04-11:
hi @jim201414 

I'm gonna add this to our task list but like I have specified many times this is a Lab project inside Arduino and when we can allocate some time to it we focus on bugs and usability.

Eventually (also based on demand) we might add features that are not of immediate need.
Thank you for letting us know, hope to make this tool more and more friendly as time goes on

ubidefeo, 2026-10-27:
little teaser

[image]