arduino/lab-micropython-editor #195: Arduino Nano ESP32 programs fail when run in the editor after updating to MicroPython version 1.26.00

siriusnerd, 2025-08-15:
After installing MicroPython 1.26.00 on ARDUINO NANO ESP32 using the MicroPython installer, it reports the following error:

> Error running dfu-util (stderr): 'dfu-util: Failed to retrieve language identifiers dfu-util: Failed to retrieve language identifiers '

Connecting the Nano ESP32 board in the Arduino Lab for MicroPython IDE generates a proper-looking connection message:

```
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OK[['boot.py', 32768, 0, 139]]
>
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 

```
But, running any program causes MicroPython to fail with seemingly non-sensical errors. For example, this program...

```
from machine import Pin
import time

LED_BUILTIN = Pin(48, Pin.OUT, value=0)   # Active high
LED_BLUE = Pin(45, Pin.OUT, value=1)      # Active low
LED_GREEN = Pin(0, Pin.OUT, value=1)      # Active low
LED_RED = Pin(46, Pin.OUT, value=0)       # Active low

while True:
  LED_BLUE.value(0)
  time.sleep(0.5)
  LED_BLUE.value(1)
  time.sleep(0.5)
```

... generates the following error message:

```
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 5
SyntaxError: invalid syntax for integer with base 10
>
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>>
``` 

Making a single change to the program by inserting a comment in line 1, like this...

```
# HelloWorld LED test program
from machine import Pin
import time

LED_BUILTIN = Pin(48, Pin.OUT, value=0)   # Active high
LED_BLUE = Pin(45, Pin.OUT, value=1)      # Active low
LED_GREEN = Pin(0, Pin.OUT, value=1)      # Active low
LED_RED = Pin(46, Pin.OUT, value=0)       # Active low

while True:
  LED_BLUE.value(0)
  time.sleep(0.5)
  LED_BLUE.value(1)
  time.sleep(0.5)
```

... generates a completely different error message:

```
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 12
SyntaxError: invalid syntax
>
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 
```

Changing the contents of the comment in line 1 also changes the reported error message! For example, changing the comment in line 1 to...

```
# HelloWorld program
```

... generates an error message that may provide a clue to a communication problem between the editor and the Arduino Nano ESP32 circuit:

```
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 9, in <module>
NameError: name 'valueOUT' isn't defined
>
MicroPython v1.26.0 on 2025-08-09; Arduino Nano ESP32 with ESP32S3
Type "help()" for more information.
>>> 

```

It appears as 'value' and 'OUT' from two separate lines of the program have been combined, omitting the content originally in-between them.

Both of these programs as well as other programs I have tried work without errors in the Arduino Labs for MicroPython editor when the ARDUINO NANO ESP32 is running MicroPython version 1.25.00. I have confirmed the errors generated are consistent using two different Arduino Nano ESP32 circuit boards running MicroPython version 1.26.00. 

I'm not sure if the cause of the errors is in MicroPython 1.26.00, the Arduino Labs MicroPython Installer program, or the Arduino Labs for MicroPython editor, but thought I should report it here for investigation since users following the steps posted in the [Arduino MicroPython 101 Course](https://docs.arduino.cc/micropython/micropython-course/course/introduction-arduino/) will certainly run into similar problems and will not be able to successfully program their circuits.

Josverl, 2025-08-15:
What host os are you running?
Is it perhaps MacOS?

siriusnerd, 2025-08-15:
> Josverl
>  left a comment
> (arduino/lab-micropython-editor#195)
> What host os are you running?
> Is it perhaps MacOS?

Yes, it’s running on MacOS – sorry, I forgot to mention that. :)

John

Josverl, 2025-08-16:
on the MicroPython repo there have been reports of issues between the MacOS and the MicroPython v1.26.0 for ESP32-S2  and ESP32-S3

- https://github.com/micropython/micropython/pull/17913
-  https://github.com/micropython/micropython/issues/17865
- https://github.com/micropython/micropython/issues/17560